CarlosGS/Cyclone-PCB-Factory #43: Guide

z3t0, 2015-04-03:
Hi, 
I am very interested in this and would definitely like to contribute to the development. Could you please give me some pointers as to where to begin as I would love to build this first and understand (to some extent) how it works so I can help to my fullest ability.

Thank you to all the contributors for this amazing project! :smile:

carlosgs, 2015-04-09:
Hello, there is no build guide yet, but a good source to follow is Joco's thread in the mail list:
https://groups.google.com/forum/#!topic/cyclone-pcb-factory/o_YzXokpLnA
Also the mail list is the best place to check for guidelines :)
Good luck with your build!!

Cheers,
Carlos

PD: Will link it here when a build guide becomes available

Obijuan, 2015-04-09:
It will be very soon :-)

z3t0, 2015-04-09:
Sounds good, if you need any help let me know. I will get started on trying to build one and make sure to document the process. How much do you think the net cost will be? Excluding the price for the 3D printed parts?

carlosgs, 2015-04-12:
The cost depends on many factors, I find it difficult to give estimates.

If you already have 3D printer hardware/electronics that can be recycled for the project, it can be very cheap. Very roughly, sourcing all the parts should be under 250€ ~(cost of a 3D printer, minus {one motor, hotend, heating/sensing elements, LCD?, etc}, plus {dremel,springs}).

Again, depending a lot on the setup you want and the sources you use :-)

Cheers,
Carlos

z3t0, 2015-04-12:
@carlosgs Okay, in that case could you make some sort of materials list I can work off, because the one on the README.md is quite confusing. Thanks

carlosgs, 2015-09-14:
Here are a couple of guides:  
BQ: http://diwo.bq.com/product/cyclone/  
[PDFs with the instructions](http://diwo.bq.com/wp-content/uploads/2015/08/Manual_ES_pdf.zip)

Also you have the build manual by Tatubias: http://reprap.org/wiki/Cyclone_PCB_Factory_Build_Manual

Cheers,
Carlos

closed 2015-09-14 by carlosgs