CarlosGS/Cyclone-PCB-Factory #51: Issues slicing models

kmckee, 2015-12-14:
I'm having a heck of a time getting Slic3r to slice a lot of these models.  I'm wonder what is recommended?

Also, if it makes sense I'd like to add this to the readme.  

Thanks; I can't wait to get this built and running.  Very cool project!

kmckee, 2015-12-14:
So, I've been able to repair the models and then slice them successfully.  I did an automatic repair using the free version of Netfabb.  

Should I send a pull request with the repaired models?

carlosgs, 2015-12-14:
Hi Kyle, I normally use Cura, which seems to incorporate the "auto repair
stl" feature, so I've never really had any problems slicing them.

But yeah it would be great to provide fixed STLs for people that uses
different gcode generators. I'd appreciate that pull request :-)

If it's not too much of a hassle, could you replace each STL in a separate
commit? That way it is easier for me to check for any difference. I'm a bit
worried the repair process may close screw holes.

>
> So, I've been able to repair the models and then slice them successfully.
I did an automatic repair using the free version of Netfabb.

kmckee, 2015-12-16:
Hi Carlos, 

I have this all ready to go, but I don't think it's possible send a PR against the v2.2 tag.  When I switch the base from a branch to a tag, github removes the "Create Pull Request" button.

Everything is in my fork here:  https://github.com/kmckee/Cyclone-PCB-Factory

First, does it make sense to repair the v2.2 stls?  If so, can you create a branch on the v2.2 tag so that I can submit the PR?

carlosgs, 2015-12-16:
Hi, the merge is done :-)
With the following commands:
'''
git remote add kmckee git@github.com:kmckee/Cyclone-PCB-Factory.git
git fetch kmckee
git merge kmckee/repairs
'''
Github seems to be clever enough to re-locate the files that were moved to different folders.

Thanks so much for taking the time to fix all of the STLs, and for logging the changes with separate commits, it helped a lot!!

closed 2015-12-16 by carlosgs

kmckee, 2015-12-16:
Awesome! 

NP; thanks for all your work on this.  I'm eagerly waiting for parts in the mail so I can put it all together :)