CarlosGS/Cyclone-PCB-Factory #52: Question about gears

czapeczek, 2016-01-26:
Hi. 

I just wanted to know (sorry if it's written somewhere - wasn't into docs really yet, I am trying to understand what I see here first) why the motors are connected to their axes with gears and  not directly?

Hope I'll hear an answer :)

Filip

Alexthefox, 2016-01-26:
Hi, I have print the same of original project. The reason is for torque of the motor. But somebody did it with direct drive, motor connect to threaded bar. Of course if change ratio of motor, need to'd change in firmware.

Alessandro

Inviato da BlueMail

czapeczek, 2016-01-26:
Okay, I assume, that by mounting a direct drive - the accuracy will be a lot better.

CarlosGS, 2016-01-26:
Gears were kept in place in order to reduce the size of the machine
Bear in mind that any backlash in the gears will be fed through the threaded rod, which is a very large reduction. You can actually have lots of backlash in the gear mechanism before it has any noticeable effect in quality

On the other hand, a word of advice. Threaded rods are usually not calibrated, which means they are not nearly as accurate as a proper industrial-grade leadscrew. If you are implementing direct drive for the sake of quality, I'd tell you to also replace the threaded rods for leadscrews

Also, please join us in the mail list: https://groups.google.com/group/cyclone-pcb-factory

Cheers,
Carlos

closed 2016-01-26 by CarlosGS

czapeczek, 2016-01-26:
Am joining right now :) 
Thanks!
The first thing I was thinking of was to changer the throds to ldscrews