CarlosGS/Cyclone-PCB-Factory #54: Is the BoM actual?

czapeczek, 2016-02-16:
I printed all the parts. It seems 2 gears not fit the M8 threaded rod and even the motor head. 
So either BoM is not actual or something is not to scale with this gears. 

I mean here gears for Z axis.

veelck, 2016-02-16:
As far as I managed to check this issue, it seems that `Cycl_Z_carriage.scad` has a bug, where it has the  `cyclone_rod_gear` call - it uses tolerance=0, where in case of X and Y axis there is tolerance set to `screwHoleTolerance`.

As for the second gear, it seems like the diameter of stepper motor rod is hardcoded inside the `standard_parts.scad`, which seems strange as I'd expect it to be configured.
It also has tolerance=0, but modifying that sets the tolerance on both holes (the motor rod as well as screwing hole, which is not really ideal).

I'd love to issue a pull request, but I'm not sure if my changes will make sense, since I'm noob in scad (started to figuring it out like 2h ago ;))

I'd love to get some feedback before I'll issue the PR.

veelck, 2016-02-17:
I'd added the `tolerance = screwHoleTolerance` for both gears, printed them and it worked like a charm, so I guess that's the problem.

CarlosGS, 2016-02-17:
@veelck Thanks for the PR! Would you mind updating the STL with the gears as well?

Interestingly enough, most complains regarding the tolerances so far have pointed out they were too large. So maybe these gears were an exception because of the bug

Thanks to both for the bug report!

veelck, 2016-02-17:
@CarlosGS Hi, thanks for taking the PR.

Can you give me some tips on how to optimally generate the STLs? Like I mentioned, I'm new to OpenCAD and I'm not sure how can I make it so it makes sense (for the gears I needed I exported them one-by-one, so for all the gears that'd be pain in the a..., hope it's not the only way...)

CarlosGS, 2016-02-17:
Unfortunately that is the case, STLs need to be manually generated for every part. What I often do, is render an STL for the full machine, then load it in Cura and choose "split into parts", finally arrange the parts into the proper alignment for 3D printing

I agree, it is tedious. But if you still have around the updated STLs, what I'd do is load [the old gear plate](https://github.com/CarlosGS/Cyclone-PCB-Factory/blob/master/Output_files/Cycl_allGears.stl) in Cura, then choose "split into parts", remove the Z gears, then drop the new STLs into Cura, and finally save the new STL plate
Sorry for the inconvenience, it would be great to have a way to automate this, indeed

CarlosGS, 2017-02-25:
Merging this issue with #57

closed 2017-02-25 by CarlosGS