CarlosGS/Cyclone-PCB-Factory #60: End stops not working with grbl

fisch3r, 2016-09-09:
Hey there,

first all thanks for your great work! 

Today I finished my build and everything is working except the end stops. I use Ramps 1.4 + Arduino Mega 2560 in combination with grbl in order to use bCnC. When I start the homing sequence the y-axis crashes into the end stop. The end stop lights up but the motor does not stop. 

I connected the x-axis end stop to x-min and the y-axis end stop to y-min. See also the attached image. 
[image]
The switches look like [these](http://g01.a.alicdn.com/kf/HTB1BD8eJpXXXXc5XXXXq6xXFXXXg/1PCS-High-Quality-Mechanical-Endstop-For-Reprap-ramps-1-4-3D-printer-With-independent-packing.jpg).

Is there a command to check the end stop status, like M119? I did not yet try marlin so far.

CarlosGS, 2016-09-12:
Hi, did you try wiping the EEPROM and flashing back GRBL?
I don't recall the command to test endstops :S 
The best check would be to measure the voltage level at each RAMPS endstop pin with a multimeter.

lasicmk2, 2016-10-17:
Also, please check while running bCNC, if you push endstop by hand bCNC generate "hard limit" alarm, if not find "command" window in lower right corner of bCNC window ant type "$21=1" - it will activate hard limits. Next try above sequence once more.

CarlosGS, 2017-02-25:
Closing this issue as it refers to firmware and software. If still persists, please submit it again to the most adequate of these:
https://github.com/CarlosGS/grblForCyclone/issues
https://github.com/vlachoudis/bCNC/issues
I hope you did manage to solve it! :-)

closed 2017-02-25 by CarlosGS