CarlosGS/Cyclone-PCB-Factory #63: grbl 1.1

MoonshineSG, 2016-12-28:
any plans to update to the new version of [grbl](https://github.com/gnea/grbl/) for RAMPS/Cyclone ?

CarlosGS, 2016-12-28:
At the moment I'm quite happy with this version, it has worked very well so
far.

So sorry but I don't plan on updating it any longer.

<https://github.com/CarlosGS/Cyclone-PCB-Factory/issues/63>, or mute the
thread

MoonshineSG, 2016-12-28:
Yes, it does work quite well. Some unsupported commands, but nothing that can't be worked around.
I was just wondering on the longer term plans. Second sentence answers that.thanks.

Great work, btw.

MoonshineSG, 2016-12-28:
oh... one of the main reasons why 1.1 is interesting is the laser support... just saying...

CarlosGS, 2017-02-25:
Maybe if I build a laser cutter in the future! I'm still waiting for high power laser diodes to drop in price a little bit.

CarlosGS, 2017-02-25:
I've moved this issue to the GRBLforCyclone repo: https://github.com/CarlosGS/grblForCyclone/issues/1

closed 2017-02-25 by CarlosGS