CarlosGS/Cyclone-PCB-Factory #66: Motor doesnt fit.

mcparyk, 2017-02-25:
I have printed a z-spindle cariagge, but my motor collides with hole for threaded rod. I am using NEMA 17 from microcon. Please fix it. This is the last thing before finishing my machine.

CarlosGS, 2017-02-25:
Hi, please check out the assembly guide for the Z axis; the rod is not supposed to go all the way through.

https://github.com/CarlosGS/Cyclone-PCB-Factory/wiki/Assembly-of-the-Z-axis#7-mount-the-z-threaded-rod-in-place (see section 8, image 2)

Let me know if this solves the problem :)

mcparyk, 2017-02-25:
Oh sorry, I placed it wrong... I am so stupid..

closed 2017-02-25 by mcparyk

CarlosGS, 2017-02-25:
No, I agree that part of the assembly is confusing, many other people had the issue as well. The Z axis would benefit from a re-design.
Oh and don't forget to say hello in [the forum](https://groups.google.com/forum/#!forum/cyclone-pcb-factory) so we can see your build! :-)