CarlosGS/Cyclone-PCB-Factory #67: Large bearings

mariustupac, 2017-05-02:
Hello! I wanna use some rexroth bearings and they have 16mm outer diameter, can someone point me to right direction on what to modify and where in the openscad files?

kejcbjn, 2017-09-21:
tggg

kejcbjn, 2017-09-21:
Hello! I wanna use some rexroth bearings and they have 16mm outer diameter, can someone point me to right direction on what to modify and where in the openscad files?

CarlosGS, 2017-09-21:
Hi, sorry for the delayed response. The bearings are defined here:
https://github.com/CarlosGS/Cyclone-PCB-Factory/blob/master/Source_files/configs/Default_config.h#L99
(List of bearing types: https://github.com/CarlosGS/Cyclone-PCB-Factory/blob/c945f72dcd6b102434cc157c840425447c59051f/Source_files/libs/linear_bearing.scad#L21 )