CarlosGS/Cyclone-PCB-Factory #71: Editable Drawing Files

FelipeBezFontana, 2017-09-13:
Hi,

I'm thinking of building a Cyclone with some changes in mechanicals parts. Can anybody send me the drawings in *.SLDPRT format?

Thank You!

CarlosGS, 2017-09-13:
Sorry, I don't know that format. I hope someone else can help!

FelipeBezFontana, 2017-09-13:
This format is from SolidWorks. Do you have any other format editable by SolidWorks or other program?

CarlosGS, 2017-09-13:
The machine is designed with OpenScad, all of the .scad sources are available. I don't think SolidWorks can import this format yet :(

FelipeBezFontana, 2017-09-13:
Sure, I get it. Thanks for the help and congratulations for the great work! :)

closed 2017-09-13 by FelipeBezFontana

CarlosGS, 2017-09-13:
There are also Step files, maybe those are better for editing rather than the STLs
https://github.com/CarlosGS/Cyclone-PCB-Factory/tree/master/bq_additions/Production/STEP

FelipeBezFontana, 2017-10-07:
Hi, I tried to edit the *.stp file in SolidWorks today and I had a good result. Thanks for helping me. ;)

CarlosGS, 2017-10-07:
That's awesome!! I hope you can modify it to fit your needs