CarlosGS/Cyclone-PCB-Factory #74: BoM redirects to Wiki main page

brnl, 2018-01-10:
Hi,

I'm curious if this project is still alive. I'm trying to research how feasible it is to build the Cyclone 2 but the (English) documentation seems outdated.

One problem is the BoM (https://github.com/carlosgs/Cyclone-PCB-Factory/wiki/Bill-of-materials) redirects to the Wiki main page (https://github.com/carlosgs/Cyclone-PCB-Factory/wiki)

Also, the [suppliers page](https://github.com/carlosgs/Cyclone-PCB-Factory/wiki/List-of-suppliers) contains only dead links.

Is the project still supported or am I better off buying a ready-made kit?

Thanks for your time!

CarlosGS, 2018-01-10:
Hi! I've fixed the link to the BoM in the main wiki page.
However, I think you are right: nowadays it is better (and cheaper) to purchase a generic CNC kit.
Cheers,
Carlos

brnl, 2018-01-10:
Hi @CarlosGS,

Thank you for your honest reply. I love your work, though. Was an awesome initiative!

Cheers,
Bram

closed 2018-01-10 by brnl

CarlosGS, 2018-01-11:
Thanks, and happy milling! :-)