CarlosGS/Cyclone-PCB-Factory #77: Selling the Cyclone PCB Factory

ShaneOberloier, 2018-09-02:
Hi Carlos -- Sorry for opening a GitHub issue, but it was the only way I could find to contact you - I realize the machine is CCbySA, but I wanted to double check - I am planning to modify this machine and begin selling it on one of my websites. Are you okay with this? Of course I will link back to your work and attribute you and your fellow contributors.

Thanks!
-Shane

Obijuan, 2018-09-03:
Hi Shane!
I am sure that Carlos will answer soon, but in the meantime, I will answer for him: you can shell it for sure! The great thing about the open licences as CCbySA is that you do not have to ask for permission. You can use Cyclone for any purpose (including commercial)

But in any case, It is great to know you like the project and you are planning to improve it and shell it. That's the purpose of the open projects! :-)

Best regars, Juan

CarlosGS, 2018-09-04:
100% agree with Juan, go for it! Looking forward to see your improvements :-)

PS: There is a [Cyclone mail list](https://groups.google.com/forum/#!forum/cyclone-pcb-factory) that is still reasonably active, you can connect with the group there. You'll also find other derivations of the design that may be useful in your design process.