CarlosGS/Cyclone-PCB-Factory #78: Could I use a dremel multipro 395 with this?

radiomaniac, 2018-09-09:
Hi, sorry this is not an issue but I don't know how to write to Carlos so I'm hoping that maybe if I post here he will read this

I want to make this project but I have a dremel multipro (model 395) could it be possible to use it with this project? Would it be a matter of designing another z carriage or is it just not possible and I have to buy another dremel? 

Thanks

electrocbd, 2018-09-09:
I have the same dremel model, and I modified the scad source code to
generate the correct z carriage.

Here is the corresponding diff:

index 9c8c49d..218ee9d 100644
--- a/Source_files/configs/Default_config.h
+++ b/Source_files/configs/Default_config.h
@@ -174,10 +174,10 @@
                spindle_holder_bolt_diam = 6;
                
        // --- Spindle Motor ---
-               spindle_motor_diam_top = 51.3;
+               spindle_motor_diam_top = 54.3;
                spindle_motor_diam_top_smaller = 47.5;
                spindle_motor_diam = 47.5;
-               spindle_motor_sidelen = 32;
+               spindle_motor_sidelen = 34;
                spindle_holder_thickness = 8;
                spindle_motor_length = 90;

Anyway, this model is no longer produced, and the resulting z carriage
lacks rigidity (it is also the case with other dremel models)

Good strategy seems to

1 Buy a modern dremel or a spindle
2 Redesign the z carriage

Easier to write than to do ;-)

Regards
electrocbd

radiomaniac, 2018-09-09:
Thanks for the reply. I think I will just sell my dremel and tools as I don't use them and then buy a better dremel or a spindle to make this thing.

CarlosGS, 2018-09-09:
Electrocbd is right on point.
And it is wiser to use a proper DC spindle, the Dremel-type tools are very noisy and get very annoying at some point. My machine still has a Dremel so I wouldn't know which spindle to recommend you.
Please drop by the [Ggroups forum](https://groups.google.com/forum/#!forum/cyclone-pcb-factory), there's much more people and support.
Cheers,
Carlos

radiomaniac, 2018-09-09:
Thanks, I was considering to buy a spindle and I was starting to think where to find the most appropriate one for the cyclone. I will search there, thanks

radiomaniac, 2018-09-10:
By the way, I'm also from Spain, carlos do you know where I can buy the springs? Thanks

CarlosGS, 2018-09-10:
Pasate por el [foro en español](https://groups.google.com/forum/#!forum/cyclone-pcb-factory-es), usa el buscador y si no encuentras lo que necesitas, adelante y manda un correo por allí :-)

radiomaniac, 2018-09-10:
No sabía que hubiera foro en español vaya, que bien. Gracias, buscaré por ahí. Busqué en el inglés pero solo encontré un mensaje preguntando por el tema de los muelles en el que respondías que no sabías de ningun sitio fuera de españa donde conseguirlos. Intentaré buscar en el foro español . Gracias