CarlosGS/Cyclone-PCB-Factory #79: Regarding Springs

moliata, 2019-01-01:
Hello,
Where can I buy springs with the parameters required for Cyclone PCB Factory. Maybe someone can make them for me? I would gladly pay for that.
Best regards,
Ben.

Helicon007, 2019-01-02:
Hi Ben,

there are no special parameters for the springs. I bought them at a standard Hardware Store/Home depot. They has to fit the threaded rods and should have enough tension to push the sliding bed against the thread, thats all. The springs just reducing the tolerance or movement of the nuts on the thread. 

Regards, Helicon

moliata, 2019-01-02:
Hello,
thank you for your response. I am more into hardware and software side, so the physical side of the Cyclone PCB Factory is hard on me. If I correctly understood, these springs should be okay? https://www.aliexpress.com/item/3D-printer-accessories-Ultimaker-2-UM2-spring-fine-print-platform-edging-1-2x15-mm-for-UM2/32818065797.html
Best regards,
Ben
P.S: those springs have 6 turns, that shouldn't be a problem.

Helicon007, 2019-01-02:
Hi Ben,

for 8mm rods they might be to tight. If 9mm is really the outer diameter you end up with 6.6mm inner. If your rods are 6mm they are perfect. In gererall you got the point, those kind of springs are fine for the Cyclone.

Regards, Helicon

moliata, 2019-01-02:
Okay, thanks! So I should get springs which have ~8.5mm inner diameter?
Best regards,
Ben.

Helicon007, 2019-01-02:
Anything bigger than 8, I don‘t know the standard values. The spring has to slide without stucking at the thread. I guess 8.2 - 8.5mm would be fine.

moliata, 2019-01-02:
Awesome, so these should be perfect: https://s.click.aliexpress.com/e/bjOfqQ7q
These follow parameters as well -> 1mm wire diameter, 11mm outer diameter, 15mm length and from what I see in the photo (correct me, if I'm wrong), these have closed end type and 5 turns.
Best regards,
Ben.

closed 2019-01-03 by moliata