CarlosGS/Cyclone-PCB-Factory #80: Height for the work and main bases

moliata, 2019-01-30:
Hello,
what should be the optimal height for the work base (as well as main)? At the moment, I am able to purchase PlexiGlass sheets with a height between 1-12mm (except 7mm and 9mm).
Best regards,
Ben.

CarlosGS, 2019-01-30:
Hi, I would recommend using wood 15mm thick. It will be easier to machine and also to fix the working materials to the base.

Otherwise the thicker the better since it will be less sensitive to vibrations.

moliata, 2019-01-30:
@CarlosGS thank you! I will be using wood for the main base, but PlexiGlass for the work base. How thick should the work base be.

Helicon007, 2019-01-30:
Hi Ben, 

I use a 12mm plexi/acryl plate as base, does the job quiet well. 

Regards, Helicon

moliata, 2019-01-30:
Isn't that too thick for the work base (milling area)?

moliata, 2019-01-30:
@CarlosGS regarding vibrations, I will be using 4pcs Rubber Feet from Adafruit, so the base won't be touching the table itself.

CarlosGS, 2019-01-30:
Good idea!

moliata, 2019-01-30:
@CarlosGS in any case, I should still go with a 15mm thick wood base, if I'm using rubber feet? Also, 12mm thick PlexiGlass for the work base (milling area)? I'm in particular looking at having the same thickness main and work as BQ's Cyclone PCB Factory. Lastly, for the "wire management tube", I am looking at a braided sleeves for cables, what size should they be to fit everything in?

closed 2019-02-05 by moliata

CarlosGS, 2019-02-05:
Sorry, I don't remember the exact diameter and don't have the machine at hand to measure it :(

moliata, 2019-02-05:
@CarlosGS when you will have the machine somewhere around, please send the diameter over here, thanks!

CarlosGS, 2019-02-07:
I see my version is not exactly the one from bq, for instance it has a different cable management tube. In my case it is a ~15mm acrylic pipe recycled from some other project.

moliata, 2019-02-07:
Thank you anyways! It seems, it's a 4mm wire management tube.

CarlosGS, 2019-02-09:
Please use the Cyclone Google groups forum for questions, instead of GitHub
issues

moliata, 2019-02-09:
@CarlosGS sorry! It doesn't seem to be active, so I made issues