CarlosGS/Cyclone-PCB-Factory #82: Mega3D Spacer

moliata, 2019-02-07:
Hello,
where can I find the DXF file for acrylic spacer (Y axis, step 12d) used for the Mega3D Board?
Best regards,
Ben.
P.S: @CarlosGS don't you have a file somewhere around? I really need it! Maybe BQ has them, if so, I could send them an email...

CarlosGS, 2019-02-07:
Hi, it should have been here: https://github.com/bq/Cyclone-PCB-Factory/tree/master/bq_additions/Proto/CAD
But they did not upload that particular dxf, probably difficult to get it back from them.

moliata, 2019-02-07:
@CarlosGS do you have the acrylic part yourself (not the file, but the actual part)?

CarlosGS, 2019-02-07:
Nope, in my machine the board is screwed to the base directly. I believe it was designed by @PedroDeOro, maybe he still has the original dxf :-)

moliata, 2019-02-07:
@CarlosGS thanks, I sent an email to Pedro, if he shares it with me, I'll post a PR, thanks!

moliata, 2019-02-12:
@CarlosGS doesn't seem he is going to respond any time soon.

closed 2020-06-21 by moliata