CarlosGS/Cyclone-PCB-Factory #83: Light metal milling

moliata, 2019-02-08:
Hello,
Is Cyclone PCB Factory okay for milling light metal like aluminum?
Best regards,
Ben.
P.S: I am using Dremel 200 as a spindle tool.

alfadex, 2019-02-08:
I don't think so

CarlosGS, 2019-02-08:
Me neither, I think it is better to use a CNC with a truly rigid structure.

closed 2019-02-08 by moliata

moliata, 2019-02-17:
@CarlosGS it seems you have milled stainless steel (https://groups.google.com/d/msg/cyclone-pcb-factory/K_wEco3CUsA/csFbfF2danwJ) which has more strength than the aluminum. I am not expecting any high-end results with clean edges, I can sand it after the milling. Regarding vibrations, I am using an 8mm thick black acrylic sheet with rubber feet. I would also use 1/8" 2mm ball nose endmill bit.

CarlosGS, 2019-02-19:
Hi Ben, have a look at other versions of Cyclone with a more rigid structure.
I don't have much experience milling metals, but remember it as a pain.