CarlosGS/Cyclone-PCB-Factory #84: Where can I get that gauge from?

jelkose, 2019-05-22:
Hi,
this is not really an issue, but I have a serious question.

In this excample picture 
(https://github.com/CarlosGS/Cyclone-PCB-Factory/commits/master/Media/Cyclone_v2.0_smallBoard2.jpg)
a really cool hole size gauge (and scale) is used.

Can anyone over here tell me what this thing is called and where I can get it?

Regards,

Jelko :-)

CarlosGS, 2019-05-22:
Hi, that awesome part was designed by Pedro de Oro and was custom made for
the company bq.

They used a water jet cutter and a bending machine.

I'm afraid I don't know any supplier for it!

jelkose, 2019-06-15:
Hmm that's sad. 
But the process sounds quite cool!

Thanks for the info!

closed 2019-06-15 by jelkose